farm-ng/tractor #14: full screen landscape web app

ethanrublee, 2020-04-25:
I've mounted my phone to the joystick in landscape mode.

Would be nice if the app could go full screen (with no address bar), and be optimized for good in landscape.

On android, I can click the chrome settings and add a launcher easily to the home screen, and according to this guide, the app when launched from the home screen can come up in full screen mode:
https://developers.google.com/web/fundamentals/native-hardware/fullscreen

[image]

[image]

mabitter, 2020-04-27:
@ethanrublee - I will have some cycles today (& more later in the week) and am thinking to take a bite into this and see how much I can chew up and spit back.  Following your lead, i'm looking at '[screenful](https://github.com/sindresorhus/screenfull.js)' as a solution.  Check out the [demo](https://sindresorhus.com/screenfull.js) link if you haven't already, the unicorn fullscreen is pretty awesome.  Hollar if you think I should bail out on this particular path of implementation, otherwise i'll fork and create branch 'frontend-screenfull' and see how well I can start to lay things out for the primary use case of a mobile device in landscape orientation.

ethanrublee, 2020-04-27:
Sounds good. Go for it.

closed 2020-09-09 by ethanrublee