farm-ng/tractor #18: Squarespace Website Down

mabitter, 2020-04-27:
**Describe the bug**
Followed link in readme to dead link : https://farm-ng.com/tractor
Trial period expired?  Ha. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://farm-ng.com/tractor'
2. See error

**Expected behavior**
Expected some awesome webpage that is going to attract patient venture capital from investors with altruism in their hearts, and developer-farmers that want to help build a better place for our children to grow up in through more sustainable farming practices and localizing food production.

**Screenshots**
[image]

**Robot HW (please complete the following information):**
 - Embedded computer: [e.g. jetson xavier, nano, raspberry pi]
 - OS Info [e.g. output of ``uname -irsv``]
 - HW version or other info about the setup [e.g. 22]

**Desktop (please complete the following information):**
 - iOS
 - Chrome

ethanrublee, 2020-04-27:
Thanks.  Yeah its not quite ready to launch :)

ethanrublee, 2020-05-08:
We'll track this issue else where.  Coming soon.

closed 2020-05-08 by ethanrublee