farm-ng/tractor #21: Specify where to discuss product specification / roadmap

isherman, 2020-05-19:
How are healthy OSS communities collaborating on product discussions, design docs, and roadmaps these days? Maybe we can brainstorm some ideas here.

a. Github Issues

b. A `farm-ng/product` repository with Markdown files. 

c. [Github Discussions](https://github.com/zeit/now/discussions/3874) (in beta)

d. [Github Team Discussions](https://help.github.com/en/github/building-a-strong-community/about-team-discussions)

e. File PRs against a Github Wiki
- Github wikis aren't set up by default to support forking/PRs, but you can jump through some hoops to set it up. 
- For example, Microsoft's [Typescript wiki](https://github.com/Microsoft/TypeScript/wiki) is managed this way (via a Github action).

f. A non-Github collaboration tool

ethanrublee, 2020-05-19:
Do you have any preferences?

My feeling is that pull requests are too onerous to make it an effective
collaboration tool on design/product discoveries.

Wiki is great, except it makes it hard to add comments or ask for
clarification.  Wiki could be good for "released" versions of
design/roadmap.

Issues seem to allow for use in github projects (
https://github.com/farm-ng/tractor/projects) for tracking things that are
related.  They do support mark down, and drag and drop of image files which
can be useful to discussion, can be linked to in code, etc.

Google Docs can be a nice collaboration tool, but maybe not so much for an
open source community.  Github gives us trace-ability and is well connected
with the implementation.

Suggested flow:

   - When starting an idea, create an issue with a special tag, like
   "ideation" or "germination".
   - Treat a "germination" issue as a sounding board for design.  Link pull
   requests or wikipages in the issue to encode the implementation.  Once the
   issue has sprouted, and has a life of its own in the implementation, close
   the issue after transcribing the summary or conclusions into a markdown/RST
   under the farm-ng/tractor repository.  markdown files are automatically
   rendered as a github page here (https://docs.farm-ng.com/tractor/) and
   this would support pull requests.
   - this flow could also be implemented using the github discussion, as
   they look similar to issues.

On the doc/markdown side of things, maybe a few templates would be useful
for product designs, or systems.  I appreciate the ROS REP and python PIP
system for documentation.  github pages could also easily be replaced with
sphinx and RST if this were convenient.

isherman, 2020-05-26:
- Agreed that opening a PR is probably too heavyweight for making a suggestion. What I like about a PR is that you can have threaded discussions about sections of the document, which is helpful for any in-depth discussion.

- I also like the idea of tracking design decisions made over the lifetime of the project, even if they were made without a design doc, or without much discussion. For example, specifying the standard units that the project uses. There are some software development processes that do this in a [heavyweight way](https://adr.github.io/); I'm picturing just a `README`-like file(s) to start. So new contributors don't have to dig through old issues.

- The Wiki seems less good than the repository for this kind of content on almost all fronts. You don't get to discuss the design in a PR, you lose portability / unified history, you can't include changes to the documents as part of larger change, and Github integrations are just not as good with the Wiki. I find wikis on Github are often kind of stale and forgotten.

- I like the REPs and PEPs too. Maybe there's a lightweight version that that feels appropriate for a much less mature project. One example I found of a lightweight enhancement proposals is in the [kubernetes/minikube project](https://github.com/kubernetes/minikube/tree/master/enhancements).

Slightly modified suggested flow:
-  When starting an idea, create an issue with a special tag, like `Enhancement`, `Feature`, or `Idea`.
- Treat a Enhancement / Feature / Idea issue as a sounding board for design.  Use linked pull requests in the issue to reference any design documents (as .md), prototype code, or the implementation.  Once the issue has sprouted, and has a life of its own in an implementation PR, transcribe the summary or conclusions of the design discussion into a .md file in the `design` directory of the `farm-ng/tractor` repository, and include those notes in the implementation PR. Use Github's [Linked Issues](https://help.github.com/en/github/managing-your-work-on-github/linking-a-pull-request-to-an-issue) feature to close the issue when the implementation PR is merged. Markdown files are automatically rendered as a github page here (https://docs.farm-ng.com/tractor/).

We could try this workflow with this issue! 🐢

ethanrublee, 2020-05-26:
+1 to the flow.

Ok, so one question is how to use the wiki if at all, as I agree with your
assessment, perhaps we'll just wait and see how the wiki turns out to be
useful.

PR #23 (2020-05-26): Document enhancement proposal process, design docs
Closes #21
closed 2020-05-26 by ethanrublee