farm-ng/tractor #24: Design Topic: Serialization and Time Series logging

ethanrublee, 2020-06-04:
TL;DR;  Create a design doc around the use of protobuf in the codebase.  Use protobuf for logging of events from the robot, leaning towards proio https://github.com/proio-org/proio over time series databases or rosbag1/2

We're leaning towards leveraging protobuf for message type definition and serialization, as well as API descriptions and server/client stub generation.  A bit of rationalization would be nice in a design doc for the choice of protobuf.

Another topic to discuss is how use protobufs in the code, in APIs and in relation to database model definitions.  I personally lean towards direct use of protobufs in APIs when possible, with conversion to language/library specific types as necessary, such as torch tensors.

Common types, patterns  and directory should be documented, such as math types.

Style and linting - what are we using currently?

Build system integration - is Uber's buf interesting? https://github.com/bufbuild/buf

I'd like to log data on the tractor, to ease the development process, to allow integration testing with real data, and playback and debug.  Similar to ROS bags... however ROS1 bags don't support protobuf.  Maybe worth exploring ROS2 bags which support protobuf potentially (https://github.com/ros2/design/blob/f69fbbd11848e3dd6866b71a158a1902e31e92f1/articles/rosbags.md)  ROS bags bring in quite a bit of legacy dependencies into the codebase.  ROS2 bags are new, and might be worth the complexity.  Benefits of ROS bags would be use of 3d debug tools from ROS, however these are not web-native, and may require special computer skills.  When possible, would like debug tooling to be accessible to both customers (farmers using the tractor) and developers.

In the past, have implemented delimited io streams of binary protobufs successfully for logging and playback of heterogeneous protobuf messages (https://developers.google.com/protocol-buffers/docs/techniques).  Basic binary format, with delimiter with serialized blob size followed by the serialized protobuf, maybe some more header info like time and message type.  prio might be an interesting existing implementation https://arxiv.org/abs/1812.03967, https://github.com/proio-org/proio

I wonder if time series databases (like sqlite style db files) might be appropriate for this. https://capablerobot.com/blog/2018/2018-05-09-wheel-reinvention/  It seems like time series databases tend to be implemented around numeric types and have limited or no support for binary blobs such as images, tensors, poses.

Another thought is to use the mp4 (or other media format) to embed synchronized data along with the camera image data, for a good example see parrot drones (https://developer.parrot.com/docs/pdraw/metadata.html).  The advantage of this approach would be great native support for video streams, and associated compression, with all other data being relatively small (gps values, VESC state, etc) and easy to side car with the video frames. https://en.wikipedia.org/wiki/MPEG-4_Part_14#Data_streams

ethanrublee, 2020-06-04:
So I've been experimenting with proio in the branch #26 

Here is a log, recorded by driving the tractor around the field.
[farm-ng-2020-06-04-15-30-29.zip](https://github.com/farm-ng/tractor/files/4733081/farm-ng-2020-06-04-15-30-29.zip)

Its self describing so if you grab the proio tools with 
```
go get github.com/proio-org/go-proio/tools/...
```

And then unzip that `.proio` file and try:
```
$ proio-summary farm-ng-2020-06-04-15-30-29.proio 
Number of LZMA buckets: 0
Number of LZ4 buckets: 0
Number of GZIP buckets: 29
Number of uncompressed buckets: 0
Number of events: 162870
Number of FileDescriptorProtos: 5
```

Or output of proio-ls
```
========== EVENT 162858 ==========
---------- TAG: right_motor/state ----------
ID: 2
unknown type: farm_ng_proto.tractor.v1.MotorControllerState
---------- TAG: stamp ----------
ID: 1
unknown type: google.protobuf.Timestamp
========== EVENT 162859 ==========
---------- TAG: rtk/solution ----------
ID: 2
unknown type: farm_ng_proto.tractor.v1.RtkSolution
---------- TAG: stamp ----------
ID: 1
unknown type: google.protobuf.Timestamp
========== EVENT 162860 ==========
---------- TAG: left_motor/state ----------
ID: 2
unknown type: farm_ng_proto.tractor.v1.MotorControllerState
---------- TAG: stamp ----------
ID: 1
unknown type: google.protobuf.Timestamp
```

I'm playing around with a component oriented design for the logging, in that each component of the system is responsible for its own logging with proio.  This means that when you want to log something new, you go to the class that is responsible for the thing you want to log, grab a handle to the application's proio logger, create an event, and push it to the writer. 

```python
plog = farm_ng.proio_utils.get_proio_logger() # can put this up at top of file in global file scope
event = plog.make_event({'%s/state' % self.name: state_msg}, stamp=stamp)
plog.writer().push(event)
```

I also found that the caching is a bit extreme with proio before it writes it to disk.  So i've played with just periodically (once per minute) flushing the log by placing the following snippet in the main application.
```
plog = farm_ng.proio_utils.get_proio_logger()
  _ = Periodic(60, event_loop, lambda: plog.writer().flush())
```

For timestamp convention, the ``make_event`` function adds as the first entry in the event a protobuf.Timestamp instance - if its not specified by the caller it grabs it from the current time.  Probably need another discussion on when timestamps refer to each event.  Thinking that each event's stamp should be as close to the measurement time as possible.  Do we also need a stamp, reflecting the latency of the event?  For example the RTK gps solution is the result of some latent solver, and refers to the gps location measured some significant (10-30ms) time in the past; the gps solution message has the stamp of the GPS measurement - should the stamp of the event in the log refer to when the RTK solution was completed, or when the GPS measurement was taken?

ethanrublee, 2020-06-04:
For video data, i'm considering the following:

- When a frame is grabbed from the camera, push an camera frame event into the log, which has the time stamp of the frame, resolution, rate, other meta data.
- Push the frame into an ffmpeg stream, to encode it into a compressed video container such as mpeg1
-  When a sufficient number of frames have been encoded to get a benefit from the compression and not have too large a protobuf message, say every 30 seconds, flush the ffmpeg stream and stuff the result in video protobuf. push the video protobuf into the proio log, with appropriate tags so that it can be cross referenced to the frame event messages.

This will require custom handling of video segments in a playback tool, but I feel like the compression is so worth the hassle, from a disk and network bandwidth perspective.

isherman, 2020-06-05:
> Style and linting - what are we using currently?
> Build system integration - is Uber's buf interesting? https://github.com/bufbuild/buf

We do currently use `buf` to build and lint all files in the `protos` directory. See https://github.com/farm-ng/tractor/blob/master/.pre-commit-config.yaml#L70-L76 and https://github.com/farm-ng/tractor/blob/master/buf.yaml. The linter settings are placeholders for now; should be specified.

> time series databases (like sqlite style db files)

Or Parquet? I think the main benefits of an approach like this is that you get columnar storage, which allows for run-length encoding and all sorts of other compression opportunities, and powerful query engines can operate directly on them. You do need to know the schema before you start writing the file (as opposed to proio, looks like you can easly register additional event types at runtime?)

> Do we also need a stamp, reflecting the latency of the event?
Might be nice if the semantics of the top-level event `timestamp` was consistent for all event types -- i.e. the time the event was logged -- but the semantics of an event's inner time-related fields were event-specific -- i.e. gps_measurement_time, rtk_solution_time. Otherwise we end up with different event types having to document what their `timestamp` means? But I'm not sure of all the consequences here, for playback systems, etc.

cbrake, 2020-07-30:
Neat project -- wish you much success.

I've been looking into using NATS in various IoT systems I'm working on. The server is very light weight, so I'm considering embedding it in a gateway (edge) app, and then connecting as a leaf node to a cloud nats server. Then it would be used for in-app, in-device, and device-device, and device-cloud messaging. The nice thing about NATS is you can easily plug into the message stream at any time. Need logging? Add a component that subscribes to topics you want to log. Want to try a new algorithm? Again, just add a component (could be a go-routine in an app, or a separate app) that subscribes to interesting data and then re-publishes processed data. NATS is touted as being very lightweight and fast, so hoping the server will work OK on devices (low end embedded Linux systems). 

A few discussions:
https://github.com/simpleiot/simpleiot/issues/62
https://github.com/simpleiot/simpleiot/issues/65

This podcast is very interesting:
https://changelog.com/gotime/130

The first time I looked at NATS, I assumed incorrectly that because it was a text protocol, payload had to be text, but it can have a binary payload -- only the framing is text.

Not saying you need something like this, but recently discovered it so thought I would mention ...

ethanrublee, 2020-08-03:
@cbrake thanks for the suggestion.  Need to look into it more.

At the moment, i've been playing with in process log writing to disk with proio.  But have started to run into logging bottlenecks due to some of the limitations of python concurrency.  So might be worth considering logging messages by sending over some IPC mechanism to a dedicated logging service.

I'm also finding the need to separate a few of our components into separate processes to deal with other bottlenecks; so having a centralized logging server that can aggregate logs from the motor control process and a tracking camera process would be great.

On a side note, on the jetson nano, i've been experimenting with gstreamer and the accelerated encoding it provides out of the box on jetson, and its pretty amazing what it can do with video compression in terms of low CPU utilization; and has built in support for udp/tcp/webrtc/etc... endpoints so might be worth incorporating this into the design - would be nice to be able to log output from gstreamer and potentially send video to a web client via webrtc.

For example, the robot can stream directly to youtube live stream with a simple gstreamer pipeline.  Here is an example recording of a youtube live stream: https://youtu.be/ONpMyScaHUo

```
        youtube_key='YOU_KEY_HERE'
        cmd=" ".join(
            ("gst-launch-1.0 fdsrc fd=0 !",
             "videoparse width=%d height=%d framerate=10/1 format=gray8 !"%(intrinsics.width, intrinsics.height),
             "videoconvert !",
             "omxh264enc control-rate=1 bitrate=10000000 EnableTwopassCBR=1 !",
             "video/x-h264, stream-format=byte-stream !",
             "h264parse !",
             "mux. audiotestsrc wave=silence freq=200 !",
             "voaacenc !",
             "aacparse !",
             "queue !",
             "mux. flvmux streamable=true name=mux !",
             "rtmpsink location=\"rtmp://a.rtmp.youtube.com/live2/x/%s app=live2\""%youtube_key
            ))
```

And here is an example that i've gotten to work for very low latency streaming from a web camera attached to robot to LAN connected browser with %7 cpu utilization:
https://gitlab.freedesktop.org/gstreamer/gst-examples/-/blob/master/webrtc/sendonly/webrtc-unidirectional-h264.c

Would love to build on top of this for video support.  webrtc also has support for realtime data channels.

I've evaluated aiortc as well, however it does not have support for hw accelerated video and i get large (growing) latency and CPU utilization (150%) when trying to use it for the webcam example: https://github.com/aiortc/aiortc/tree/main/examples/webcam

cbrake, 2020-08-03:
Nice work with the video!

yeah, one of the nice things about NATS or similar is the logging does not sit in the primary flow of data -- it is simply another consumer of the stream, so it does not slow things down.

Another interesting option for Go is https://watermill.io (have not used it yet). I really like Go in these types of projects. The concurrency is great, but what really helps me is the reliability and ease of deployment (statically linked runtime and dependencies). Image processing/ML etc are nice in Python though.