farm-ng/tractor #25: pre-commit hook fails eslint error - react

ethanrublee, 2020-06-04:
I'm trying to use the pre-commit hooks for linting, I keep hitting this error and not sure how to resolve it.  I've run yarn in the app/frontend directory and npm says i have react installed.  Still learning my way around javascript stuff.

```
eslint...................................................................Failed
- hook id: eslint
- exit code: 2

TypeError: text.trimEnd is not a function
Occurred while linting /home/rublee/code/tractor/app/frontend/src/components/Tractor.tsx:1
    at attachComments (/home/rublee/.cache/pre-commit/reposgzm8kev/node_env-system/lib/node_modules/prettier/index.js:14771:60)
    at coreFormat (/home/rublee/.cache/pre-commit/reposgzm8kev/node_env-system/lib/node_modules/prettier/index.js:14798:23)
    at format (/home/rublee/.cache/pre-commit/reposgzm8kev/node_env-system/lib/node_modules/prettier/index.js:15019:75)
    at formatWithCursor (/home/rublee/.cache/pre-commit/reposgzm8kev/node_env-system/lib/node_modules/prettier/index.js:15035:12)
    at args (/home/rublee/.cache/pre-commit/reposgzm8kev/node_env-system/lib/node_modules/prettier/index.js:51620:12)
    at Object.format (/home/rublee/.cache/pre-commit/reposgzm8kev/node_env-system/lib/node_modules/prettier/index.js:51640:12)
    at Program (/home/rublee/.cache/pre-commit/reposgzm8kev/node_env-system/lib/node_modules/eslint-plugin-prettier/eslint-plugin-prettier.js:241:41)
    at listeners.(anonymous function).forEach.listener (/home/rublee/.cache/pre-commit/reposgzm8kev/node_env-system/lib/node_modules/eslint/lib/linter/safe-emitter.js:45:58)
    at Array.forEach (<anonymous>)
    at Object.emit (/home/rublee/.cache/pre-commit/reposgzm8kev/node_env-system/lib/node_modules/eslint/lib/linter/safe-emitter.js:45:38)
```

isherman, 2020-06-05:
Looking into it....

isherman, 2020-06-05:
I'm unable to reproduce on my machine with a clean:
```
git clone --recurse-submodules
./env.sh
cd app/frontend; yarn install; cd ../..
virtualenv venv
. venv/bin/activate
pip install -r requirements.txt -r requirements-dev.txt
pre-commit run --all-files
```

Googling the error message, it looks like the issue is related to running a version of node < 10.0:
https://github.com/prettier/eslint-plugin-prettier/issues/278. `text.trimEnd` was introduced in node v10.0

I tried changing the default node version on my machine to 8.10 but unfortunately still couldn't reproduce. I still suspect node version is at the root of it though, partly because I know your machine had an older version of node installed by default. Maybe there's some caching behavior happening around pre-commit's invocation of `node` that I don't understand.

ethanrublee, 2020-06-05:
Thanks.

Wish there was a way to run the pre-commit hooks in a docker so that it was repeatable.  I'm not familiar with node enough yet to understand how to have the proper toolchain installed.

ethanrublee, 2020-06-05:
```
$ node --version
v12.16.3
```

ethanrublee, 2020-06-05:
Interesting - so I upgraded my node from an apt installed version, following a guide similar to this one: http://geekstuff.org/2018/11/04/how-to-install-node-ubuntu-18-04/

Then I just tried purging the apt installed version ``apt purge nodejs``

And it looks like for some reason the node used by pre-commit doesn't look in the correct place
```
/home/rublee/.cache/pre-commit/reposdeu1nks/node_env-system/bin/node: line 5: /usr/bin/nodejs: No such file or directory
```

```
(venv) f-g:rublee@farmws01:~/code/tractor$ which node
/usr/local/bin/node
(venv) f-g:rublee@farmws01:~/code/tractor$ node -v
v12.18.0
```

After uninstalling the apt version of node, i removed the pre-commit cache and re-ran, and it seems "fixed".

```
(venv) f-g:rublee@farmws01:~/code/tractor$ rm -rf ~/.cache/pre-commit/
(venv) f-g:rublee@farmws01:~/code/tractor$ pre-commit run --all-files
```

isherman, 2020-06-05:
Great, makes (some) sense. I wasn't able to get as far as figuring out which `node` binary pre-commit was invoking. 

pre-commit does let you run docker hooks! we already use it actually:
```
-   repo: local
    hooks:
    -   id: proto-lint
        name: proto-lint
        language: docker_image
        pass_filenames: false
        entry: docker.io/bufbuild/buf:latest check lint
```

we could migrate this eslint stuff to docker if it continues to be a headache

isherman, 2020-06-05:
Anything else before close?

ethanrublee, 2020-06-05:
I think it can be closed... but maybe we should have a bit of a plan for making dev env reproducible.  Seems like a broader topic - venv, node, etc...

closed 2020-06-05 by ethanrublee