farm-ng/tractor #27: phone - tractor - mesh network topology

ethanrublee, 2020-06-04:
We're planning to use a phone as the main UI device in the field. One simple option is to have good mesh coverage, and have the phone and tractor connected to the mesh network.  This may result in high latency for sending commands between the phone and tractor depending on the quality of the mesh network.  Other options would be to have a point to point connection between the tractor and the phone, such as over bluetooth, or a secondary wifi card on the tractor (https://abdusyarif.net/2017/12/05/share-internet-connection-via-bluetooth-on-ubuntu/). Another possibility is to put a mesh wifi AC on the tractor (Unifi Mesh AC $90), and then ethernet wire the tractor brain, and then wifi connectivity from phone to tractor should be great, and get high power Tx/Rx to mesh backbones (Not sure how well mesh network works with a rover based access point :).  

At the farm-ng research farm, we tried both cell phone wifi and the tractor connected to the 2g mesh network, and get > 1 mbps to both devices at 200 meters away from the access points in our open field, but found that placing body between the phone and the access point at this distance negatively affects bandwidth and ping times.

ethanrublee, 2020-06-04:
Blueman supports network share from linux node to phone over wifi (https://abdusyarif.net/2017/12/05/share-internet-connection-via-bluetooth-on-ubuntu/)  but may be a pain for users to figure out the phone network side of things...

ethanrublee, 2020-06-04:
For adding external antenna to raspberry pi -
https://dorkbotpdx.org/blog/wramsdell/external_antenna_modifications_for_the_raspberry_pi_3/

isherman, 2020-06-05:
For ease-of-use I like the idea of a point-to-point connection, with a Wifi AP + a big honking antenna on-tractor.

And the option to participate in an externally managed (mesh, or otherwise) network for the user who already has that infrastructure.

But I don't know what kind of power requirements that imposes on the tractor. Or, if the base station is already a required bit of infrastructure, whether it makes sense to centralize power-hungry radios and large antennae there. Seems like we might benefit from some outdoor networking expertise.

ethanrublee, 2020-06-05:
For the outdoor network, I'm thinking the unifi mesh network is required for operation of the tractor v1, and will be the dedicated backbone for the first trials.  Its unlikely that farms have this existing infrastructure.

https://unifi-mesh.ui.com/

The access points can be managed by small docker-compose setup from here https://github.com/jacobalberty/unifi-docker

Outdoors in fields, the larger mesh-pro APs provide amazing coverage with 2g.

We already require a link back to the RTK base station for RTK updates - this could be over LoRA or some other long range network with low bit rate - but the benefits of high bandwidth wifi, and prevalence of devices that can talk 2g/5g seems to make this the ideal choice for the first deployments.  1 pro AP in the center of a field can support ~20 acre square.

ethanrublee, 2020-09-09:
This is a bit outdated - we're going with default mode is the tractor has a wifi AP, #62

closed 2020-09-09 by ethanrublee