farm-ng/tractor #60: address video streaming hiccups on LAN

isherman, 2020-09-04:
We see the video pause briefly. This seems to correspond to packetLoss on the inbound-rtp video stream to the browser, which can be seen here:

[image]

I believe a single lost packet will pause the video until the next keyframe arrives.
 
An incomplete list of areas to explore: 
- RTP Buffering in the webrtc proxy ([increasing the ReadBufferSize](https://github.com/farm-ng/tractor/pull/33/commits/5b01035d2d70967925c121fde86d5669011ec06a) helped a lot)
- Removing RTP, interfacing with gstreamer queue instead (w/ consequences for development workflow -- can no longer run proxy on workstation)
- Adjusting gstreamer knobs (increasing keyframe frequency to improve recovery from dropped packets, etc)

In short, more tuning to do with chrome://webrtc-internals

ethanrublee, 2020-09-04:
I wonder if the kernel recv buffer could help as well, or maybe a red haring.  Was reading about problems caused by this here: https://lcm-proj.github.io/multicast_setup.html

On my machine this is the current setting:
```
$ sysctl  net.core.rmem_max
net.core.rmem_max = 212992
```

Its curious that the gstreamer udpsrc doesn't suffer from this performance issue.  Is it also possible that the udp recv handling is just too slow to keep up in the go code? 

If RTP is not handled well in go, you could use a gstreamer pipeline embedded in go to recv the RTP stream with udpsrc, and then have an appsink or fdsink. https://github.com/notedit/gstreamer-go

backup options

- webrtc has been released for the jetson platform fully optimized (https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/hardware_acceleration_in_webrtc.html)

- gstreamer also supports webrtc (and only needs the signalling handled externally).  Cool example from cinder (https://discourse.libcinder.org/t/cinder-gstwebrtc/1573)

isherman, 2020-09-04:
`net.core.rmem_max` definitely seems relevant -- however a quick test yesterday didn't show any visible difference.
It's a good point about comparing it to gstreamer rtpsrc performance -- I will try to do some experiments there.
And yeah, agreed, I think we could try the appsink or fdsink approach, without RTP (or even UDP) in the middle.