farm-ng/tractor #70: event_bus is slow when we have a few processes sending lots of messages

ethanrublee, 2020-09-07:
Implement message filtering on send side, based on processes which seem interested in a given topic.

When running the tracking_camera process, which sends 200hz pose + imu data, all processes receive this data.  On the nano, the decoding negatively impacts the steering and tractor process (doubling the CPU utilization...).

isherman, 2020-10-14:
Some profiling of the `tractor` service:

### Without T265 pose streaming
- 63% of the time spent in `epoll` (higher the better)

[image]

- The other 40% spent mostly in IPC send and SE3 / numpy
[image]

### With T265 pose streaming
- 30% spent in `epoll`

[image]

- The other 70% spent roughly 50/50 in IPC send/receive

[image]

ethanrublee, 2020-10-14:
Cool thanks for profiling!

PR #136 (2020-10-16): Add subscriptions to IPC
Closes #70
closed 2020-10-16 by isherman