farm-ng/tractor #78: farm_ng.wifi exception

mabitter, 2020-09-09:
Tried to follow instructions under /tractor/jetson/readme.md, and bust. 

**Screenshots**
```
(env) farmer@farmer:~/tractor$ sudo ./env.sh python -m farm_ng.wifi ap
AP Password (at least 8 characters): 
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/farmer/tractor/python/farm_ng/wifi.py", line 184, in <module>
    WifiCLI()
  File "/home/farmer/tractor/python/farm_ng/wifi.py", line 161, in __init__
    getattr(self, args.command)()
  File "/home/farmer/tractor/python/farm_ng/wifi.py", line 167, in ap
    enable_ap()
  File "/home/farmer/tractor/python/farm_ng/wifi.py", line 126, in enable_ap
    connection = get_ap_connection()
  File "/home/farmer/tractor/python/farm_ng/wifi.py", line 61, in get_ap_connection
    NetworkManager.Settings.AddConnection(get_ap_config(password))
  File "<string>", line 4, in AddConnection
  File "/home/farmer/tractor/env/lib/python3.6/site-packages/dbus/proxies.py", line 147, in __call__
    **keywords)
  File "/home/farmer/tractor/env/lib/python3.6/site-packages/dbus/connection.py", line 653, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.NetworkManager.Settings.Connection.InvalidProperty: 802-11-wireless-security.psk: property is invalid
```

git sha
```
(env) farmer@farmer:~/tractor$ git rev-parse HEAD
30095eac45d9417510f0ff5afd38c8c0112fa4f0
```

isherman, 2020-09-09:
Were you prompted for a password? Did you ensure the password was at least 8 characters long?

mabitter, 2020-09-09:
Yes we were prompted for a password, "farm-ng" is however NOT 8 characters
long... so we didn't follow directions.  Doh.

isherman, 2020-09-09:
Please close if it works with a longer password. Thx!

closed 2020-09-14 by isherman