farm-ng/tractor #81: Dangerous behavior after boot and pairing of joystick

mabitter, 2020-09-10:
**Describe the bug**
After boot, and after pairing the controller, if the deadman or 'L2' is pressed BEFORE any actuation of the joysticks the robot will take off under power as if someone is commanding full speed ahead.  Sometimes only one wheel is activated, other times both, sometimes it doesn't do it.  It is always in the forward direction...  

**To Reproduce**
Steps to reproduce the behavior:
1. reboot the robot (hard or soft) 
2. pair the joystick by pressing the ps4 button as usual 
3. press the deadman (L2)
4. run for your life! 

**Expected behavior**
It should not move. 

**Temporary workaround by MM** 
After boot, and after pairing the controller, actuate the forward and reverse joystick to its full extents, as well as the L/R turn joystick to its full extents.  Then press the deadman and operate as usual.

ethanrublee, 2020-09-10:
I think this is a quirk related to the brand of controller; i can't reproduce it with the Sony PS4 controller.  I'll try with a generic one.

If you could enable verbose debugging in the joystick.py module (Set logging to DEBUG https://github.com/farm-ng/tractor/blob/master/python/farm_ng/joystick.py#L26)

The stop all the robot processes, and run python -m farm_ng.joystick, pair the joystick, move the thumb sticks, and past the output here, that would be helpful.

mabitter, 2022-07-05:
deprecated

closed 2022-07-05 by mabitter