farm-ng/tractor #104: Support generic "programs"

isherman, 2020-09-23:
# Motivation

We're starting to see examples of small procedural-style programs we'd like to run on the tractor.

These programs:

- Have an explicit beginning and end
- Interact with services running on the tractor (e.g. the camera service)
- Interact with persistent storage
- Require some frontend for visualization and user interaction

An example would be running a sensor-to-sensor calibration. 
- This is not a service; it has an explicit beginning and end
- It requires the orchestration of, at minimum, the camera and calibration solver
- It must be able to store the calibration result for use in the future
- It requires a frontend to allow the user to visualize progress and make decisions about how to proceed

# Architecture

With the IPC infrastructure we already have in place, the first three requirements are not difficult to meet.
- The operating system will start and stop programs for us. 
- Our services already run as daemons. 
- Persistent storage can be provided by the filesystem, or a database.

There are some details to manage around, for example, ensuring that only one program is running at a time.

The frontend requirement makes things a bit more complicated. 

The frontend requirements means we now need to author a UI in addition to the program logic, and typically the state of the UI is highly coupled with the state of the program.

Ideally, the UI associated with a new program is relatively self-contained from the rest of the tractor frontend. Think more 'App Store' than Adobe Photoshop.

We'd like to find an opinionated, repeatable approach to building these programs and their UIs, for consistency, maintainability, and low marginal cost of new programs.

## Web architecture

Consider a common architecture for a web application.

Assume the backend is a monolith that provides two kinds of persistent storage, a relational database and a static asset directory. Assume the frontend runs a React SPA.

Adding a new "program" typically involves:

- Adding a table to the database
- Defining a set of new HTTP endpoints to CRUD the new database table
- Extending a "client" in the frontend to interface with the new endpoints
- Extending a "store" in the frontend to represent the additional, locally-cached backend state
- Defining a set of "actions" in the frontend that define how endpoint responses should update the store

In a microservices architecture a couple details change (e.g. an additional database instead of a new table) but the broad strokes are the same.

## Proposed architecture

The proposed architecture responds to a few differences between our use case and a traditional web application:

- We don't have multiple clients interacting with the robot at once.
- We don't have relational data to manage (at least not yet).
- Existing communication infrastructure is built around the eventbus, not RPC / HTTP.
- Programs and their frontends should be largely self-contained, rather than extending a monolith.

In the proposed architecture:

- A "program" on the robot is a (hopefully simple) Python program, making use of async/await to interact with asynchronous APIs.
- All communication between programs, services, and UI is via the eventbus.
- All persistent storage is via flat files on the filesystem.
  - We're mainly concerned with storing non-relational blobs, some of them large
  - This means the user can easily browse and edit the "database"
  - Because we only expect one client interacting with the robot at once, we don't need support for concurrent writes.
- A "program" will manage all of its state (or more precisely, the state that should be reflected in the UI) in a single Protobuf-defined message.
  - Any updates to the program's state should re-publish the entire program state on the eventbus
  - The frontend will one-way synchronize the program state to its store, which in the React one-way data-binding paradigm will update the view
- A "program" may define additional messages that the UI may publish on the eventbus as a means of updating the program state

### 1. Run sensor-to-sensor calibration

As a user I want to:

- Start capturing frames, see frames arrive as they are captured.
- Stop capturing frames.
- Choose which rigs to use for calibration, or abort the calibration.
- Start calibration, see calibration progress as it crunches numbers.
- See the calibration result, and if it's successful, decide whether to save it.

```python
# update_state applies a diff to the program state object and publishes the latest complete program state on the event bus
update_state({state: 'INIT'})

# await_with_timeout awaits an asynchronous operation. if the operation doesn't complete
# by the end of the timeout period, it invokes update_state({state: 'ERROR', errorMessage: <error>})
# and exits
await_with_timeout(eventbus.all_ready(['logger', 'calibrator', 'camera']), 5000)

update_state({state: 'READY'})

# wait for a user interaction from the frontend (or a CLI, log playback, etc.)
# (don't do anything with the received message)
await_with_timeout(eventbus.on('StartCapture', None), 60000)

update_state({state: 'CAPTURING'})

eventbus.on('AprilTagRig', lambda d: update_state({detected_rigs: state.detected_rigs + [r]}))
eventbus.send(CameraCommand(command: {record_start: {mode: MODE_EVERY_APRILTAG_FRAME}}))

# wait for a user interaction from the frontend
await_with_timeout(eventbus.on('StopCapture', None), 60000)

update_state({state: 'CAPTURE_COMPLETE'})

# maybe a good practice to deregister?
eventbus.off('AprilTagDetection')
eventbus.send(CameraCommand(command: {record_stop}))

# wait for a user interaction from the frontend
await_with_timeout(eventbus.on('SelectedRigsForCalibration', lambda r: update_state({selected_rigs: r})), 60000)

update_state({state: 'CALIBRATING'})

eventbus.on('CalibratorStatus', lambda s: s.solver_status === CONVERGED ? update_state({result: s} : update_state({progress: s})))
eventbus.send(CalibratorCommand(apriltag_rig_start: {state.selected_rigs}))

# wait for a user interaction from the frontend
var save_result = false
await_with_timeout(eventbus.on('SaveCalibrationResult', lambda s: save_result = s , 60000))
if save_result:
  with file.open(env.get(BLOBSTORE_PATH) + "/" + state.run_id as f):
    f.write(...)

update_state({state: 'DONE'})
exit()
```

#### Questions

- Does this procedural-style, with handlers, work as programs get more complex?
  - e.g. what about a cancellation event that could arrive at any time?
- I still feel a bit funny about the amount of fire-and-forget there is for what feel intuitively to be synchronous APIs (start recording, stop recording). But I might just need to do some more reading about event-driven architectures. For example the pattern described above with the program emitting its full state seems to map onto the concept of "Event-Carried State Transfer".

### 2. Build Tag Library

As a user I want to:

- Start capturing frames, see frames and tagIds arrive as they are captured.
- Stop capturing frames.
- Fill out a form of <tagId, tagDimension>, and give this tagLibrary a name.
- Submit the form and assume the tagLibrary is saved.

```
TODO
```

#### Questions

This one feels a little funny to me as I start to ask, is this program the way we edit existing tag libraries? Browse tag libraries? As soon as we're in the world of primarily performing CRUD operations on persistent objects, the 'program' paradigm doesn't feel quite right.

### 3. Run sensor-to-link calibration

As a user I want to:

- Start capturing frames, see frames arrive as they are captured
- Start the 'robot dance', see wheel odometry arrive as it is captured.
- Stop capturing frames.
- Choose which rigs to use for calibration, or abort the calibration.
- Start calibration, see calibration progress as it crunches numbers.
- See the calibration result, and if it's successful, decide whether to save it.

```
TODO
```

ethanrublee, 2020-09-24:
random thoughts in response:

I'd like to push back on having to collect the state into a single protobuf for the app.  The state from each service is already sent on the event bus, and the presentation of this state can be controlled easily in the frontend code (which is tightly coupled with the app).  For example, it seems onerous to have to copy the detected_rigs into state, when the frontend would already have access to this message directly from the source.  Might be nice if the frontend for an app has some minimal command/status message protocol specific to the app, and then can compose events from other services to help with presentation.

Some good reading on patterns for this type of application domain: Real-Time Systems: Design Principles for Distributed Embedded Applications,  Hermann Kopetz (probably I need to make another pass at reading it, its been a few years).  I take a bit of inspiration for how the canbus event and time triggered schemes work in embedded systems.  There are good error handling paradigms described in the book as well.

Perhaps we should avoid the DSL part of this for a bit until we implement an app or two.  It would be nice to consider writing the applications in c++ as well.

Suggest focus on ironing out the protobuf conventions, and event bus message command/status protocols, and how the services/app interact, as we de-risk the event bus design - then we'll either know we should switch to (g)RPC, or have enough examples to create a nice syntax.

I've been playing with asyncio queue.

```python
async def get_message(event_queue, name_pattern, message_type):
    program = re.compile(name_pattern)
    message = message_type()
    while True:
        event = await event_queue.get()
        if program.match(event) is None:
            continue
        if event.data.Unpack(message):
            return message

# usage 
        
async def some_task(bus, event_queue):
   bus.send(make_event("logger/command", log_start_command))
   status = await get_message(event_queue, "logger/status", LoggingStatus)

```

Note here that we're passing a predicate to wait for an event that matches what we expect.  This could throw an exception or return None on error.

isherman, 2020-09-24:
> I'd like to push back on having to collect the state into a single protobuf for the app. The state from each service is already sent on the event bus, and the presentation of this state can be [...]

sure, i can start there. i was attempting to lower the complexity of the frontend development by encapsulating the full program state in a single message. without that, each frontend app will have to provide its own code to pull events off the bus and merge them into what's effectively a materialized view for the frontend. that is more idiomatic to frontend application development, it's just one more pattern to learn for contributors who may not have frontend development experience.

> I've been playing with asyncio queue.

this is useful, thank you

ethanrublee, 2020-09-25:
So some more prototyping in the #83 branch.  This is what the calibration program ends up looking like.

```python
async def start_recording(event_queue):
    await wait_for_services(event_bus, ['ipc-logger', 'tracking-camera', 'calibrator'])
    event_bus.send(
        make_event(
            'logger/command',
            LoggingCommand(
                record_start=dict(
                    archive_path=args.name,
                    mode=LoggingCommand.RecordStart.MODE_ALL_MESSAGES,
                ),
            ),
        ),
    )

    await wait_for_predicate(
        event_queue, 'logger/status', LoggingStatus,
        lambda status: status.HasField('recording'),
    )

    event_bus.send(
        make_event(
            'tracking_camera/command',
            TrackingCameraCommand(record_start=dict(mode=TrackingCameraCommand.RecordStart.MODE_APRILTAG_STABLE)),
        ),
    )

async def stop_recording(event_queue):
    event_bus.send(
        make_event(
            'tracking_camera/command',
            TrackingCameraCommand(record_stop={}),
        ),
    )

    event_bus.send(
        make_event(
            'logger/command',
            LoggingCommand(record_stop={}),
        ),
    )

    await wait_for_predicate(
        event_queue, 'logger/status', LoggingStatus,
        lambda status: status.HasField('stopped'),
    )

async def calibrate(event_queue):

    event_bus.send(
        make_event(
            'calibrator/command',
            CalibratorCommand(apriltag_rig_start=dict(tag_ids=args.tag_ids)),
        ),
    )

    await wait_for_predicate(
        event_queue, 'calibrator/status', CalibratorStatus,
        lambda status: (
            status.HasField('apriltag_rig') and
            status.apriltag_rig.num_frames >= args.num_frames
        ),
    )

    event_bus.send(
        make_event(
            'calibrator/command',
            CalibratorCommand(solve={}),
        ),
    )

    return await wait_for_predicate(
        event_queue, 'calibrator/status', CalibratorStatus,
        lambda status: (
            status.HasField('apriltag_rig') and
            status.apriltag_rig.HasField('rig_model_resource') and
            status.apriltag_rig.finished
        ),
    )

with EventBusQueue(event_bus) as event_queue:
    try:
        event_loop = asyncio.get_event_loop()
        event_loop.run_until_complete(start_recording(event_queue))
        event_loop.run_until_complete(calibrate(event_queue))
    finally:
        event_loop.run_until_complete(stop_recording(event_queue))
```

Still a bit verbose, but it properly stops the recording if ctrl-c is pressed, and synchronizes the processes using the event bus.

jinfwhuang, 2020-09-28:
> All persistent storage is via flat files on the filesystem.

You could consider a locally hosted S3. https://github.com/minio/minio/

The good:
-  If later on you decide that your storage should be in the cloud, the only thing that needs to change is changing `--endpoint-url=127.0.0.1:9000` to something like `s3.us-east-1.amazonaws.com`. There are clients that will for both google cloud storage and S3 compatible, e.g. boto3. 
- If you starting to have large files that you want to keep, you can turn on the cloud option with no work.
- You would not worry about concurrent write/read issues to the file systems.

However, a blob storage adds another layer of indirection your "db". Inspecting the file is through a cli tool, e.g. `aws s3 ls`, instead of looking at a plain file system.

isherman, 2020-09-30:
> You could consider a locally hosted S3. https://github.com/minio/minio/

Thanks @jinfwhuang, that's a good reference. I was also looking at Go CDK's [blob package](https://gocloud.dev/howto/blob/) for an API over local and remote file storage. I think for now we're going to assume our resource-generating code is "well-behaved" and doesn't contend for writing the same file.

isherman, 2020-09-30:
> I'd like to push back on having to collect the state into a single protobuf for the app. The state from each service is already sent on the event bus, and the presentation of this state can be [...]

The [first implementation](https://github.com/farm-ng/tractor/pull/107/files#diff-0bbfd0cb4b6fb0e6e864d10f96f27254) took a bit of different shape than I was imagining, thanks in part to your feedback @ethanrublee, and you know, actually writing the code. I abandoned the idea of a "program" being a Python program with a standard structure, and instead just support exec'ing any binary. I abandoned the idea of "materializing" a program state in the program for the UI to consume. I'm slowly getting more comfortable with the eventbus paradigm.

The repeatable approach to building program UI is achieved, for now, by rending a "streaming log" of the subset of events on the eventbus associated with the program:

[image]

In this approach, the program-specific work just becomes implementing visualizers for any new event types the program introduces.

Now working on prototyping a solution to deal with user input in a semi general way.

closed 2020-09-30 by isherman