farm-ng/tractor #106: Address known issues in oscilloscope

isherman, 2020-09-24:
- [ ] No support for streaming images yet (i.e. remote resource archive).
- [ ] Images are still visible from previous frames in overlay view, even if resource field is not populated
- [ ] In "Overlay" and "Grid" views, newest data is on the last page rather than the first.
- [ ] Grid view is jumpy when some messages are populated and others are not; may need fixed sizes
- [ ] JSON displays enum values; seeing the key name would be more useful.
- [ ] Error switching from Streaming to Log view if the selected datatype is not present in the log.
- [ ] Check whether `autorun` is necessary in `app/frontend/src/components/scope/visualizers/ApriltagDetectionsVisualizer.tsx`
- [ ] Investigate filename length limit in .tar file format

ethanrublee, 2020-09-27:
The Plot function, for example the visualization for the Steering command, increases the vertical size continuously, making the plot move vertically, and hard to use.

isherman, 2020-09-30:
> The Plot function, for example the visualization for the Steering command, increases the vertical size continuously, making the plot move vertically, and hard to use.

This was a regression 😞, will fix.