farm-ng/tractor #154: Building the repo on Ubuntu 18.04 desktop

collinbrake, 2020-11-18:
@isherman 

Following the development guide is going well. I few minor issues have cropped up, mostly due to user error:

- Did not have virtualenv installed the first time I ran `make`
- The second time I tried the build, the error message said that the bootstrap recipe failed, due to some files already existing in  `/tmp`
  - resolved by running `sudo rm -r /tmp/node_exporter-1.0.1.linux-amd64* /tmp/prometheus-2.21.0.linux-amd64*`
- The last time I tried the build, this error resulted:
```code
     ERROR: Command errored out with exit status 1:
     command: /home/collin/farm-ng/tractor/env/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-p2vNjK/websockets/setup.py'"'"'; __file__='"'"'/tmp/pip-install-p2vNjK/websockets/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-mv_RWW
         cwd: /tmp/pip-install-p2vNjK/websockets/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-p2vNjK/websockets/setup.py", line 1, in <module>
        import pathlib
    ImportError: No module named pathlib
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

- After installing `python-pathlib` the same error resulted.
  - The process that produces this error is "Collecting websockets"
  - The first time (before installing python-pathlib) it said "Downloading websockets-5.0.1.tar.gz" but now says "Using cached websockets-5.0.1.tar.gz"

isherman, 2020-11-18:
Thanks @collinbrake.

I've been able to reproduce all the issues you've run into so far.
Looks like the latest issue you're stuck on is a python2 vs. python3 issue.

I'm working on a branch: https://github.com/farm-ng/tractor/compare/master...isherman:bootstrap-debug
On that branch you can see the `dev.Dockerfile` I'm using to replicate a clean ubuntu 18.04 install.
Once I have that branch fully building from scratch I'll update again here.

(We're talking about moving to a dockerized development environment, which will help significantly with the reproducibility of the dev environment. Thanks for bearing with us.)

isherman, 2020-11-18:
@collinbrake I got past the issues you described above, and as far as I could in my Dockerized environment (basically up until our build scripts invoke Docker...).

Could you try making the changes to `bootstrap-apt.sh` and `bootstrap-venv.sh` in https://github.com/farm-ng/tractor/compare/master...isherman:bootstrap-debug (or checkout that branch), and let me know where that gets you?

collinbrake, 2020-11-18:
@isherman Sure, I'll do that!

collinbrake, 2020-11-18:
@isherman The updated scripts get me past all of the issues above!

Now I am running into this:

```code
Successfully installed cython-0.29.21 dbus-python-1.2.16 future-0.18.2 liegroups-1.1.0 linuxfd-1.4.4 numpy-1.19.4 prometheus-client-0.9.0 protobuf-3.14.0 python-networkmanager-2.1 six-1.15.0 tornado-6.1 websockets-8.1
make[1]: Leaving directory '/home/collin/farm-ng/tractor'
make third_party
make[1]: Entering directory '/home/collin/farm-ng/tractor'
cd third_party && ./install.sh
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 163, in _run_module_as_main
    mod_name, _Error)
  File "/usr/lib/python2.7/runpy.py", line 119, in _get_module_details
    code = loader.get_code(mod_name)
  File "/usr/lib/python2.7/pkgutil.py", line 281, in get_code
    self.code = compile(source, self.filename, 'exec')
  File "/home/collin/farm-ng/tractor/python/farm_ng/config.py", line 16
    def _in2m(inches: float) -> float:
                    ^
SyntaxError: invalid syntax
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 163, in _run_module_as_main
    mod_name, _Error)
  File "/usr/lib/python2.7/runpy.py", line 119, in _get_module_details
    code = loader.get_code(mod_name)
  File "/usr/lib/python2.7/pkgutil.py", line 281, in get_code
    self.code = compile(source, self.filename, 'exec')
  File "/home/collin/farm-ng/tractor/python/farm_ng/config.py", line 16
    def _in2m(inches: float) -> float:
                    ^
SyntaxError: invalid syntax
```

collinbrake, 2020-11-18:
Python 2.7 is installed on my system. I am not sure whether that is the installation that the `tractor` code is using or not. It looks like the error resulted from function type annotations which are not supported in Python 2.

isherman, 2020-11-18:
Could you try sourcing `setup.bash`? This should activate the virtualenv, which should contain the correct version of python.
```
# You can either:
. setup.bash
# or:
./env.sh

# then 
make
```

isherman, 2020-11-21:
Hi @collinbrake just a heads-up we just landed a large refactoring change on `master`. Hopefully not too disruptive to you. Happy to jump on a chat or call next week to bang through any outstanding dev environment issues!

collinbrake, 2020-11-22:
Thanks @isherman! I will make sure to source that setup file next time I try.

isherman, 2020-11-24:
From @mabitter :

> I couldn't find the github issue mentioned, but wanted to follow up as I tried to follow along this morning and got blocked at a different bootstrap issue :  

```
Successfully built python-networkmanager liegroups
+ FARM_NG_GOPATH=/home/vaquero/tractor/env/go
+ mkdir -p /home/vaquero/tractor/env/go
+ export GOPATH=/home/vaquero/tractor/env/go:
+ GOPATH=/home/vaquero/tractor/env/go:
+ go get -u github.com/golang/protobuf/protoc-gen-go
/home/vaquero/tractor/bootstrap-venv.sh: line 24: go: command not found
Makefile:15: recipe for target 'bootstrap' failed
make[1]: *** [bootstrap] Error 127
make[1]: Leaving directory '/home/vaquero/tractor'
Makefile:7: recipe for target 'all' failed
make: *** [all] Error 2
```

isherman, 2020-11-24:
It's supposed to work like this:
```
make bootstrap
|- bootstrap.sh
 |- bootstrap-apt.sh
  |- installs go
 |- bootstrap-venv.sh
  |- installs go libraries into virtualenv

. setup.bash (or ./env.sh)
|- sources virtualenv (making go and python libraries available)

```

collinbrake, 2020-11-25:
@isherman I pulled master and then merged in your fixes from [master...isherman:bootstrap-debug](https://github.com/farm-ng/tractor/compare/master...isherman:bootstrap-debug). After sourcing `setup.bash` and attempting a build, this error triggered:

```code
CMake Error: The source directory "/home/collin/farm-ng/tractor/third_party/opencv" does not appear to contain CMakeLists.txt.                                                                                     
Specify --help for usage, or press the help button on the CMake GUI.                                                                                                                                               
make[2]: Entering directory '/home/collin/farm-ng/tractor/third_party/build-opencv'                                                                                                                                
make[2]: *** No targets specified and no makefile found.  Stop.                                                                                                                                                    
make[2]: Leaving directory '/home/collin/farm-ng/tractor/third_party/build-opencv'                                                                                                                                 
make[2]: Entering directory '/home/collin/farm-ng/tractor/third_party/build-opencv'                                                                                                                                
make[2]: *** No rule to make target 'install'.  Stop.                                                                                                                                                              
make[2]: Leaving directory '/home/collin/farm-ng/tractor/third_party/build-opencv'                                                                                                                                 
CMake Error: The source directory "/home/collin/farm-ng/tractor/third_party/ceres-solver" does not appear to contain CMakeLists.txt.
```

Do your fixes from bootstrap-debug need merged to the master branch?

Thanks,
Collin

@mabitter

collinbrake, 2020-11-25:
@isherman Trying to understand the code base better:

From looking at the  `CMakeLists` files in the repo, it looks like executable(s) are generated in four modules:

- core
- tractor
- calibration
- perception

For basic functionality with a usb camera, which of these modules will we be using? I assume that the code is fairly interconnected, i.e. the tractor module might rely on the core module?

mabitter, 2020-11-25:
Tried to run ``./bootstrap.sh`` and it seems to depend on sourcing setup.bash (for the golang path environment variables).  Suggest that bootstrap.sh can be run with a clean environment ( so it sources setup.bash or sets the environment variables itself).

Error bellow:
```

+ FARM_NG_GOPATH=/home/farmer/tractor/env/go
+ mkdir -p /home/farmer/tractor/env/go
+ export GOPATH=/home/farmer/tractor/env/go:
+ GOPATH=/home/farmer/tractor/env/go:
+ go get -u github.com/golang/protobuf/protoc-gen-go
/home/farmer/tractor/bootstrap-venv.sh: line 24: go: command not found

```

isherman, 2020-11-25:
> `CMake Error: The source directory "/home/collin/farm-ng/tractor/third_party/opencv" does not appear to contain CMakeLists.txt.`

^ This looks like an issue with not cloning submodules (`git clone --recursive` or after clone, `git submodule update --init --recursive`)

> For basic functionality with a usb camera, which of these modules will we be using? I assume that the code is fairly interconnected, i.e. the tractor module might rely on the core module?

Yup, there are dependencies between modules. For that use case you'd be using core (no dependencies) and perception (depends on core).

> Do your fixes from bootstrap-debug need merged to the master branch?

Probably, yes.

> Matthew's golang issue

Not sure what the root cause of this is yet.

Let's hop on a chat and try to figure this out!

PR #158 (2020-12-04): Refining Build Scripts/Docs for Clean Ubuntu 18.04 Installation
closes #154
closed 2020-12-04 by isherman